hbinderup94/E17-PROJ3-Gruppe-7 #25: Reverse Engineer Instagram

hbinderup94, 2017-10-24:
Miw

Hestehviskeren, 2017-11-09:
Instragram har ssl pinning og signatur key generation, som gør API'en meget svær at reverse engineer.

hbinderup94, 2017-11-09:
Er blevet udsat til SPRINT 4, da den tager længere tid end forventet

Hestehviskeren, 2017-11-20:
Mangle problemer med Instragram API ift. sikkerhedsforanstaltninger.
Der påbegyndes en erstatning løsning hvor der benyttes Flickr API (Instragram ligende side)

Hestehviskeren, 2017-11-23:
Blevet lavet om til FB, vi uploader istedet billeder til en Facebook Side via. Facebook API.

Hestehviskeren, 2017-11-24:
Virker og testet - brugervenlige funktioner er blevet lavet.
- Mangler noget error-handling, men dette bliver implementeret i UI.
[FB-Page-API.tar.gz](https://github.com/hbinderup94/E17-PROJ3-Gruppe-7/files/1501551/FB-Page-API.tar.gz)

closed 2017-11-24 by Hestehviskeren